mesh-iit/cad-libraries #99: Editing [DIN7984]-[correcting material]

Mick3Lozzo, 2022-06-08:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | [ECAD-MCAD]                        |
| Alias Creo    | DIN7984                   |
| IIT-Code      | not available - screw family table           |
| Applicant     | Michele Gesino, @Mick3Lozzo             |
| Supervisor    | Ugo Pattacini, @pattacini        |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <br>
This family table has not correctly set the material properties. The correct material of the component is steel Aisi 304 (asked telephonically to the supplier).

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Checked that the associated [WINGST](http://wingst.icub.iit.local/) code is up-to-date and that the manufacturer and at least one supplier are defined.

fiorisi, 2022-06-08:
> The correct material of the component is steel Aisi 304 (asked telephonically to the supplier).

The AISI 304 is generally class A@ steel. Did you as the state (annealed, hardened, etc)?
Depending on the state one can define the resistance of the material. 
See https://github.com/icub-tech-iit/cad-mechanics/wiki/Technical-specification-standards#screws-bolts-and-nuts

marcellosavoldi, 2022-06-10:
Are you speaking about these 2 screws?
[image]

The Bossard code in wingst is [BN2844](https://www.bossard.com/eshop/global-en/screws-and-bolts-with-internal-drive/hex-socket-head-cap-screws-with-low-head-partially-fully-threaded/p/2844/?category=&index=0&q=bn2844%3Arelevance#tab-features)...
the material type indicated as "A2" only...

checking in the [Wurth website](https://eshop.wuerth.it/Categorie-di-prodotti/DIN-7984-acciaio-inossidabile-A2/31143511140116.cyid/3114.cgid/it/IT/EUR/?CampaignName=SR001&VisibleSearchTerm=din7984) I found the same screws with the material Type "A2-70" ; the same of the torx screws...

Do you think we can use the same?

[image]

Mick3Lozzo, 2023-08-09:
@fiorisi @marcellosavoldi can we close this issue?

fiorisi, 2023-08-09:
> @fiorisi @marcellosavoldi can we close this issue?

Yes, let's use the A2-70.

closed 2023-10-27 by Mick3Lozzo