mesh-iit/cad-libraries #101: Connect Tech - NGX004-02 - Quark Carrier + NVIDIA® Jetson Nano™ Module + Active Thermal

salvi-mattia, 2022-06-14:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  |[ carrier ](https://connecttech.com/product/quark-carrier-nvidia-jetson-nano/) , [heatsink](https://connecttech.com/product/nvidia-jetson-nano-active-heat-sink/) , [jetson nano](https://developer.nvidia.com/embedded/jetson-nano)                |
| Project              | 4k head     |
| Alias Creo           | CONNECTTECH_NGX004-02                        |
| IIT-Code             | [if available]                         |
| Applicant            | @salvi-mattia                |
| Supervisor           | @fiorisi             |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
[image]

[image]

[image]

[image]

[image]

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

salvi-mattia, 2022-06-14:
16446 - Quark Carrier + NVIDIA® Jetson Nano™ Module + Active Thermal NGX004-02

we need to check weight and mounting while the assembly will be here

MrAndrea, 2022-07-01:
Hi @pattacini, @masa, why did you assign only the Wingst code 16446 for this group of components and not 3 different codes?

[image]

cc @marcellosavoldi

MrAndrea, 2022-07-01:
Thanks.
Now we have to replace the Nano board with the Xavier NX as written in [this comment](https://github.com/icub-tech-iit/study-icub-head/issues/15#issuecomment-1168439612), and the Xavier NX has 16487 as code. 
Do we add a new Wingst code for the new group or do we split the 3 components?

fiorisi, 2022-07-01:
Thanks @salvi-mattia for the detailed comment.

Probably at this point it would be useful to have a clear list and connection schematics using the dedicated issue template: https://github.com/icub-tech-iit/mechatronics/issues/new/choose

[image]

salvi-mattia, 2024-01-08:
let's assume from the OP that the whole code weights 46+33+1.4= 80.4g
this seems not to be coherent with [this](https://github.com/icub-tech-iit/study-icub-head/issues/6#issuecomment-1135864747)
at the moment we do not have a physical specimen to be weighed

salvi-mattia, 2024-01-08:
talking with @fiorisi fe found out that the correct weight of the here discussed boards pack
is 46+33+1.4+18 (https://www.waveshare.com/jetson-nano-module.htm) = 98.4g

salvi-mattia, 2024-01-08:
[image]

i guess we can close @fiorisi

closed 2024-01-08 by fiorisi