mesh-iit/cad-libraries #103: BASLER - 2000036061 - M12 Lens, Focus 2.1mm, Aperture 1.8 FOV 170 degree with IR cut filter, 1/3

salvi-mattia, 2022-06-28:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | file:///C:/Users/msalvi/Downloads/DG00211003000_Tech_Spec_for_SAP_2000036060_2000036061-2.pdf               |
| Project              | ErgoCub & iCub   |
| Alias Creo           | BASLER_2000036061                       |
| IIT-Code             | 16483                         |
| Applicant            | @salvi-mattia            |
| Supervisor           | @pattacini              |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [x] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [x] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [x] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [x] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [x] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [x] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [x] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

salvi-mattia, 2022-06-28:
mass check required

Lawproto, 2026-01-16:
The item exists in both libraries and WinGST and has all the boxes ticked. Not sure about the mass check but I am pretty sure we moved on from Basler lenses. So we can close.