mesh-iit/cad-libraries #105: Leopard Imaging - 2920-650-120 - M12 Lens, Focus 2.9mm, Aperture 2.0 FOV 120 degree

salvi-mattia, 2022-06-28:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [file:///C:/Users/msalvi/Downloads/2920-650-120_2.9mm_Spec-2.pdf                  |
| Project              | iCub & ergoCub     |
| Alias Creo           | LEOPARD-IMAGING_2920-650-120                         |
| IIT-Code             | [13483                        |
| Applicant            | Name Surname, @salvi-mattia               |
| Supervisor           | Name Surname, @pattacini                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

salvi-mattia, 2022-06-28:
mass check required

Lawproto, 2026-01-16:
The item is complete, even if lacking of real-life weight check. Closing as complete.