mesh-iit/cad-libraries #106: [BOE Technology] - [BOE_BF060Y8M-AJ0-7D00] - [FHD color active matrix AMOLED display, 5.99 inch]

Mick3Lozzo, 2022-07-12:
| Description          | BOE OLED flexible display 5.99" |
|:---------------------|:---------------------------------------|
| Datasheet component  |    [6 inches panel data with FPC.pdf](https://github.com/icub-tech-iit/cad-libraries/files/9094494/6.inches.panel.data.with.FPC.pdf)              |
| Project              | [ergoCub    |
| Alias Creo           | BOE_BF060Y8M-AJ0-7D00   |
| IIT-Code             | 16508                      |
| Applicant            | Michele Gesino, @Mick3Lozzo               |
| Supervisor           | Ugo Pattacini, @pattacini               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with the manufacturer and at least one supplier.

Mick3Lozzo, 2022-07-12:
We don't have an official supplier in WinGST, this item has been bought via Aliexpress.

@marcellosavoldi @fiorisi

marcellosavoldi, 2022-07-13:
I noticed that the attached document is confidential ...
Can we put this item in the public library?
although in theory it can be purchased at ALI....

Mick3Lozzo, 2022-07-13:
> I noticed that the attached document is confidential ... Can we put this item in the public library? although in theory it can be purchased at ALI....

It is downloadable via registration @marcellosavoldi. I don't know. 🤷‍♂️

marcellosavoldi, 2022-07-14:
> It is a commercial but I set the `type` parameter to `ELTR`, right?
> 
> @marcellosavoldi @fiorisi

Yes... because electronic boards usually are buying from electronic team...

Mick3Lozzo, 2023-08-09:
This component has been discontinued. Closing issue.