mesh-iit/cad-libraries #107: [Digital View] - [MIP-1000] - [OLED / LCD Controller Board]

Mick3Lozzo, 2022-07-13:
| Description          | MIPI DSI OLED / LCD Controller Board        |
|:---------------------|:---------------------------------------|
| Datasheet component  | [digital-view-product-guide-2022.pdf](https://github.com/icub-tech-iit/cad-libraries/files/9102982/digital-view-product-guide-2022.pdf), [mip-1000-brochure.pdf](https://github.com/icub-tech-iit/cad-libraries/files/9102997/mip-1000-brochure.pdf)  |
| Project              | [where will the component be used]     |
| Alias Creo           | Digitalview_MIP-1000              |
| IIT-Code             | 16509                |
| Applicant            | Michele Gesino, @Mick3Lozzo              |
| Supervisor           | Ugo Pattacini, @pattacini        |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with the manufacturer and at least one supplier.

Mick3Lozzo, 2022-07-13:
Created code in WinGST, but we don't have an official supplier, it has been bought via Alexpress.

Mick3Lozzo, 2022-07-13:
[image]

Weighted mass, 18 g.

marcellosavoldi, 2022-07-14:
Can you create a middle plane in the USB connector and flat connector to facilitate the assembly of the plug components?
what do you think?
[image]

fiorisi, 2022-07-14:
> Created code in WinGST, but we don't have an official supplier, it has been bought via Alexpress.

cc @icub-tech-iit/silo-electr

fgarini, 2022-07-14:
> > Created code in WinGST, but we don't have an official supplier, it has been bought via Alexpress.
> 
> cc @icub-tech-iit/silo-electr

May Electronica be a suitable candidate for this request of offer?
They already supplied to us the flexible display (cfr purchases#1739)

Mick3Lozzo, 2022-07-15:
> Can you create a middle plane in the USB connector and flat connector to facilitate the assembly of the plug components? what do you think?

Ok.

Mick3Lozzo, 2023-08-09:
This component has been discontinued. Closing issue.

closed 2023-08-09 by Mick3Lozzo