mesh-iit/cad-libraries #110: RS - 196-5457- Metric Nitrile 70 ShA O-Rings Ø int. 2.6mm, Ø ext. 5.4mm 1.4mm

salvi-mattia, 2022-07-14:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  |https://docs.rs-online.com/f20f/A700000006699145.pdf            |
| Project              | 4k eye     |
| Alias Creo           |RS_196-5457                      |
| IIT-Code             | [if available]                         |
| Applicant            | @salvi-mattia              |
| Supervisor           | @pattacini              |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

salvi-mattia, 2022-07-14:
as per
https://github.com/icub-tech-iit/cad-libraries/issues/108#issuecomment-1184403887

Lawproto, 2026-01-20:
Despite the related ISO3601 family table never been created, this item exists in cad-libraries and the info is complete. It is an o-ring in case we have to trace the issue back for future reference (e.g. o-ring family table creation). Closing for now.