mesh-iit/cad-libraries #114: Maxon - B7E7C938897D - Motor - DCX10S EB KL 9V + Gearhead - GPX10 A 256:1

Lawproto, 2022-07-27:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [b7e7c938897d_2.pdf](http://wingst-icub.iit.local/direct_download.php?LINK=%2F%2Fiiticubstor01.iit.local%2Ficub-tech%2Fmechatronics%2FDatasheet%2FMotors%2FMaxon%2Fb7e7c938897d_2.pdf)              |
| Project              | Mk5 hand |
| Alias Creo           | MAXON_B7E7C938897D    |
| IIT-Code             | 16582                  |
| Applicant            | Lorenzo Protopapa, @Lawproto        |
| Supervisor           | Luca Fiorio, @fiorisi          |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Lawproto, 2022-07-27:
Assigned density so that the mass is `15 g` as per datasheet.

Lawproto, 2022-08-16:
Weight of the physical object implemented.

[image]

[image]

Closing.

closed 2022-08-16 by Lawproto

fiorisi, 2022-08-17:
Thanks @Lawproto.
I wonder if the 2.5 grams of difference between the datasheet and the weight scale are due to the wires.
Considering that currently in our CAD  models we don't have cables, and considering that usually the weight of the real robot is less than the one estimated by the CAD, I would use 15 grams instead of 17.5.

What do you think?

Lawproto, 2022-08-17:
@fiorisi given a commercial component, how do we decide when to prioritise the datasheet weight over the measured one?
If a clear and easy criterion does not come mind, I prefer to stick with the simpler rule we adopted until now. Also considering the difference is small enough.