mesh-iit/cad-libraries #119: [BotBlox] - [BB-GGB-C-1] - [GigaBlox – Small GigaBit Switch]

Elio75, 2022-08-23:
| Description          | GigaBlox – Small GigaBit Switch 45x45mm - 5 ports  |
|:---------------------|:---------------------------------------|
| Datasheet component  |[ GigaBlox](https://botblox.io/products/gigablox-small-gigabit-switch)|
| Project              | ergoCub     |
| Alias Creo           | BOTBLOX_BB_GGB_C_1   |
| IIT-Code             | 16281                        |
| Applicant            | @Elio75      |
| Supervisor           | @fiorisi             |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
[GigaBlox_DataSheet_C_C.pdf](https://github.com/icub-tech-iit/cad-libraries/files/9402549/GigaBlox_DataSheet_C_C.pdf)
[GigaBlox_Rev_C.pdf](https://github.com/icub-tech-iit/cad-libraries/files/9402550/GigaBlox_Rev_C.pdf)

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Elio75, 2022-08-23:
[image]

marcellosavoldi, 2022-09-08:
@Elio75 , se poi riesci a mettere i sys "CONN_MATE" sui connettori che useremo sarebbe stupendo...

Elio75, 2022-09-08:
OKA

Elio75, 2022-09-20:
Fatto 

[image]