mesh-iit/cad-libraries #130: Carlstahl - CG077081 - Stainless steel microcable

Lawproto, 2022-11-10:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | kinda everywhere, really    |
| Alias Creo           | CG077081         |
| IIT-Code             | 4482                       |
| Applicant            | Lorenzo Protopapa, @Lawproto |
| Supervisor           | Luca Fiorio, @fiorisi     |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:
This is a bulk item, so some checkboxes do not apply.

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2022-11-10:
@Lawproto maybe you can add a ~strikethrough~ on the points of the checklist that don't apply.

Lawproto, 2022-11-10:
Committed and pushed, see commit above.
Closing.

closed 2022-11-10 by Lawproto