mesh-iit/cad-libraries #140: Magfine - MAGFINE_NDC_MAGNET - Neodymium N35 Magnets with hole

Mick3Lozzo, 2023-02-10:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [Neodymium countersunk circular magnets(https://www.magfine.it/it/user_data/neodymium_disc_countersunk.php)                  |
| Project              | ergoCub     |
| Alias Creo           | MAGFINE_NDC_MAGNET                        |
| IIT-Code             | [if available]                         |
| Applicant            | Michele Gesino, @Mick3Lozzo              |
| Supervisor           | Ugo Pattacini, @pattacini                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
 Example of instance NDC0023
 
[NDC0023_2D.pdf](https://github.com/icub-tech-iit/cad-libraries/files/10708007/NDC0023_2D.pdf)
[NDC0023_SPEC.pdf](https://github.com/icub-tech-iit/cad-libraries/files/10708009/NDC0023_SPEC.pdf)
[NDC0023_3D.pdf](https://github.com/icub-tech-iit/cad-libraries/files/10708010/NDC0023_3D.pdf)

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Mick3Lozzo, 2023-02-10:
For the material, I added the Neodymium N35 to the material library, considering the fact that all the physical properties of the material were known @icub-tech-iit/silo-mech.

The EDU-COM converter still is not working.

fiorisi, 2023-02-13:
> For the material, I added the Neodymium N35 to the material library, considering the fact that all the physical properties of the material were known @icub-tech-iit/silo-mech.
> 
> For now, the only instance I added is **17122**.

@Mick3Lozzo did you open a Pull Request?
If not please delete the material and open a PR.

Mick3Lozzo, 2023-02-13:
> @Mick3Lozzo did you open a Pull Request? If not please delete the material and open a PR.

Added @fiorisi.

Mick3Lozzo, 2023-02-13:
[image]

BTW @fiorisi, these are the magnets we are using in R1. 👆🏻

Maybe we need another \*.mtl file, or we could use the same kind of neodymium.

closed 2023-08-09 by Mick3Lozzo