mesh-iit/cad-libraries #143: Xsens - MTi-630 - IP 51 rated IMU

Mick3Lozzo, 2023-02-21:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  |    [MTi_600_series_Datasheet-2940224.pdf](https://github.com/icub-tech-iit/cad-libraries/files/10792474/MTi_600_series_Datasheet-2940224.pdf) |
| Project              | ergoCub    |
| Alias Creo           | XSENS_MTI-630           |
| IIT-Code             | [if available]                         |
| Applicant            | Michele Gesino, @Mick3Lozzo    |
| Supervisor           | Ugo Pattacini, @pattacini            |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Mick3Lozzo, 2023-02-21:
[image]

Weight: **37 g**

marcellosavoldi, 2023-02-22:
@Mick3Lozzo If you can, put a SYS Mate_conn, with the plug connector as a cosmetic in the 'PLUGS' layer as written in the [WIKI](https://github.com/icub-tech-iit/cad-libraries/wiki/Create-a-ECAD-component-in-CREO#mandatory-coordinate-system-for-plug-components).
This is useful for defining the overall dimensions.
thanks

fiorisi, 2023-02-22:
@Mick3Lozzo please add also a frame that defines the measuremetn frame. Please post also the page of the manual that defines its location, as a future reference.

Mick3Lozzo, 2023-02-22:
> @Mick3Lozzo please add also a frame that defines the measurement frame. Please post also the page of the manual that defines its location, as a future reference. Check other sensors to use the same naming convention.

@fiorisi the frame has been already added following the manual indications (pages 27-37 of the attached manual).

[image]

[image]

[image]

I also added as a hidden copy geometry a possible connector suggested in the manual, but there are a lot of possible choices, and the correct one is still to define.

[image]

fiorisi, 2023-02-22:
> @fiorisi the frame has been already added following the manual indications (pages 27-37 of the attached manual).

Great! Can you check how it is named in other sensors, like FT or boards with accelerometers.

> I also added as a hidden copy geometry a possible connector suggested in the manual, but there are a lot of possible choices, and the correct one is still to define.

Then I would remove it.

marcellosavoldi, 2023-02-23:
Hi @Mick3Lozzo , I just talked to @fiorisi  about the plug connector and explained to him the usefulness of the connector's cosmetics.
He also agrees that it can be useful to keep it in the hidden level...
If we then use another type of connector we can make an envelope with a connector with the maximum size of all or simply overlap them.... then we will decide.
is it ok for you?

Mick3Lozzo, 2023-02-23:
[image]

I was joking! 😶 I verified the IMU we bought, and in the box, there are included **exactly** the connectors I choose.

For now, I am going to maintain them, but I think there is some idea to buy a cable with an adapter. Maybe.

@marcellosavoldi @fiorisi

Mick3Lozzo, 2023-03-02:
@fiorisi @marcellosavoldi, @maggia80 said that we are going to realize a cable by ourselves and the COM Express probably has a UAART connector too, so I suppose that [this one](https://user-images.githubusercontent.com/15085666/220938991-f8008f80-6895-4b79-b204-ad3e082437c5.png) is definitely the correct one.

Mick3Lozzo, 2023-04-21:
Created WinGST code **17298**.

@fiorisi @marcellosavoldi

fiorisi, 2023-04-21:
@Mick3Lozzo once the part is done feel free to close the issue.

closed 2023-04-21 by Mick3Lozzo