mesh-iit/cad-libraries #144: ECAD - [RS75_ALUMINUM_HOUSED_RESISTOR] - [IIT-CODE] - [short description]

Andrea-Mor, 2023-02-23:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [link or attachment ] *               |       |
| Manufacturer info  | [Manufacturer name + serial number] * |       |
| Alias Creo         | [Alias or filename MCAD] \**          |       |
| MCAD Contact       | Andrea Mor @Andrea-Mor              |       |
| ECAD Contact       | Francesco Garini @fgarini               |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Andrea-Mor, 2023-02-23:
Hi @fgarini 
can you please work on those points?
[image]

fgarini, 2023-02-24:
Code: 7138 
Description: Aluminium housed wirewound resistor, 1R2, 75W, 5%
Alias: RES-AXIAL-1R2-75W-ALU-5TOL

Datasheet | [[link or attachment](http://wingst.icub.iit.local/direct_download.php?LINK=%2F%2Fiiticubstor01.iit.local%2Ficub-tech%2Fmechatronics%2FDatasheet%2FResistors%2FRS%2FRS-75+Aluminium+Housed+Resistors.pdf) ]|  
Manufacturer info | RS 1754152* |  
Alias Creo | [Alias or filename MCAD] ** |  
MCAD Contact | Andrea Mor @Andrea-Mor |  
ECAD Contact | Francesco Garini @fgarini |  
Maintainer | Marcello Savoldi, @marcellosavoldi

Andrea-Mor, 2023-02-24:
Done, closing.
cc @fiorisi

closed 2023-02-24 by Andrea-Mor