mesh-iit/cad-libraries #147: [manufacturer] - [part number] - [short description]

Elio75, 2023-04-04:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | Tubular bubble levels with frame to screw on, Form B, three sight glasses (K1158)                |
| Project              | ergoCub    |
| Alias Creo           | MISUMI_K1158_2571305                        |
| IIT-Code             | 17264                        |
| Applicant            | Elio Massa @Elio75                |
| Supervisor           | F. Larosa, @Fabrizio69               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:
[Livella_Misumi.pdf](https://github.com/icub-tech-iit/cad-libraries/files/11149042/Livella_Misumi.pdf)
[K1158.2571305](https://uk.misumi-ec.com/vona2/detail/221303941397/?KWSearch=Bolla&searchFlow=results2products)

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2023-04-04:
@Elio75 please check the issue title.

Elio75, 2023-12-13:
[image]

[image]