mesh-iit/cad-libraries #159: [Connecttech] - [XHG312] - [NVIDIA Jetson Xavier NX Active Heatsink]

Mick3Lozzo, 2023-10-20:
| Description          | NVIDIA Jetson Xavier NX Active Heatsink, 40mm fan  |
|:---------------------|:---------------------------------------|
| Datasheet component  |  [CTIM-XHG312_Manual.pdf](https://github.com/icub-tech-iit/cad-libraries/files/13053502/CTIM-XHG312_Manual.pdf) |
| Project              | ergoCub  |
| Alias Creo           | Connecttech_XHG312 |
| IIT-Code             | 16945                  |
| Applicant            | Michele Gesino @Mick3Lozzo             |
| Supervisor           | Nicolò Genesio, @Nicogene   |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

[image]

### Notes:

### Checklist:
Complete the checklist to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Nicogene, 2023-10-20:
> Weight to verify, the internet says 0.8, but what?

You can check what weght has in the iCub 3 cad since it is mounting the **Xavier NX**

Mick3Lozzo, 2023-10-23:
Weight XHG312: **57g** 
source: the web.

fiorisi, 2023-10-23:
> Weight XHG312: **57g** source: the web.

@Mick3Lozzo maybe you can add a screenshot or a link.

Mick3Lozzo, 2023-10-23:
[image]

⬆️ @fiorisi

Mick3Lozzo, 2023-10-23:
[image]

Updated mass.

marcellosavoldi, 2023-10-25:
I think it might be useful to define on the model where the electrical wires come out of...
is it by any chance possible to indicate them figuratively on the 3D model?

Mick3Lozzo, 2023-11-02:
[image]

@marcellosavoldi added cosmetic cables.

closed 2025-05-19 by Mick3Lozzo