mesh-iit/cad-libraries #160: [Connect Tech] - [NGX007] - [Boson Carrier board for NVIDIA]

Mick3Lozzo, 2023-10-20:
| Description          | Boson Carrier board for NVIDIA® Jetson Nano™ and Xavier NX or ORIN NX Module, 9-36V input |
|:---------------------|:---------------------------------------|
| Datasheet component  |  [CTIM_NGX007_Manual.pdf](https://github.com/icub-tech-iit/cad-libraries/files/13053798/CTIM_NGX007_Manual.pdf), [NGX007_3D_MODEL.zip](https://github.com/icub-tech-iit/cad-libraries/files/13053840/NGX007_3D_MODEL.zip)  |
| Project              | ergoCub |
| Alias Creo           | CONNECTTECH_NGX007  |
| IIT-Code             | 17678           |
| Applicant            | Michele Gesino, @Mick3Lozzo         |
| Supervisor           | Nicolò Genesio, @Nicogene       |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

[image]

### Notes:

### Checklist:
Complete the checklist to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

marcellosavoldi, 2023-10-20:
out of curiosity, how do you intend to handle flat cables?
in theory it will change shape depending on the assembly position.... right?

Mick3Lozzo, 2023-10-20:
[image]

Yep, but this ⬆️ is the CAD model. [This one](https://user-images.githubusercontent.com/15085666/276921515-7b9c1b0a-60c7-41dd-b98b-f6a358e2295e.png) was only a more complete picture from the nVidia website.

Mick3Lozzo, 2023-10-20:
[image]

Weight: 80 g (source: datasheet)

Mick3Lozzo, 2023-10-23:
[image]

Updated mass.

closed 2025-05-19 by Mick3Lozzo