mesh-iit/cad-libraries #164: [manufacturer] - [part number] - [short description]

Elio75, 2023-12-13:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [HP-Series-Standard-Data-181107.pdf](https://github.com/icub-tech-iit/cad-libraries/files/13662217/HP-Series-Standard-Data-181107.pdf)|
| Project              | ergoCub1.1     |
| Alias Creo           | [AMPHENOL_HP-3-MDW / 17761] - [AMPHENOL_HP-3-MDGW / 17760]|
| IIT-Code             |cod.17761 / cod.17760                      |
| Applicant            | Elio Massa @Elio75               |
| Supervisor           | F. Larosa, @Fabrizio69                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Elio75, 2023-12-13:
AMPHENOL_HP-3-MDGW  cod.17760

[image]

Elio75, 2023-12-13:
AMPHENOL_HP-3-MDW  cod.17761 weight 12gr

[image]

marcellosavoldi, 2023-12-28:
It might be useful to add a cosmetic plug as described in the[ WIKI](https://github.com/icub-tech-iit/cad-libraries/wiki/Create-a-ECAD-component-in-CREO#mandatory-coordinate-system-for-plug-components) to better verify the actual dimensions during the study.
remember the right level "PLUGS".

Elio75, 2024-01-16:
> It might be useful to add a cosmetic plug as described in the[ WIKI](https://github.com/icub-tech-iit/cad-libraries/wiki/Create-a-ECAD-component-in-CREO#mandatory-coordinate-system-for-plug-components) to better verify the actual dimensions during the study.
remember the right level "PLUGS".