mesh-iit/cad-libraries #165: Harwin - R30-6700394 - Spacer, cylindrical, int.diam. 3.2mm, through hole, h=3mm, nylon

Lawproto, 2024-01-10:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [see Wingst]                 |
| Project              | ergoCub lower arm   |
| Alias Creo           | HARWIN_R30-6700394                   |
| IIT-Code             | 4591                       |
| Applicant            | Lorenzo Protopapa, @Lawproto                 |
| Supervisor           | Luca Fiorio, @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Lawproto, 2024-01-10:
See commit above. Closing.

Lawproto, 2024-05-17:
@Mick3Lozzo it is a duplicated code. I say we just keep one of them, replace all the where used of the other and then delete it. Anyway, I suggest that the library component issue stay close since it is mostly unrelated to this problem.

Mick3Lozzo, 2024-05-20:
@Lawproto I did a check and a component with a 4591 alias does not even exist, maybe I can correct the alias of [this](https://private-user-images.githubusercontent.com/15085666/331539892-0ba7614f-bdc3-4189-be90-8a57ab9bd3ee.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxOTA1NzMsIm5iZiI6MTcxNjE5MDI3MywicGF0aCI6Ii8xNTA4NTY2Ni8zMzE1Mzk4OTItMGJhNzYxNGYtYmRjMy00MTg5LWJlOTAtOGE1N2FiOWJkM2VlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIwVDA3MzExM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQxNzIyMjA2MTQ3ZjVjNjNiYWMyNmQ0NWE3OWVjYWVkYTM2YTE0ZDg1MDNlYjI3NWQ0YjhiNzUzNDE5YzYxZTAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.rFFz0--EEvj4Ii0w37M-7OzSG_molPrIC4NL_8lYnPI) and make [this one](https://private-user-images.githubusercontent.com/15085666/331541041-3cebebad-5ef0-47db-a8a2-0ae7d9d4a161.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTYxOTA1NzMsIm5iZiI6MTcxNjE5MDI3MywicGF0aCI6Ii8xNTA4NTY2Ni8zMzE1NDEwNDEtM2NlYmViYWQtNWVmMC00N2RiLWE4YTItMGFlN2Q5ZDRhMTYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA1MjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNTIwVDA3MzExM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTViNjFhYTg2MmNlNjA1Y2JjZTY1NzY4YjlkMTEwYzMxZDMyZWQxNzA4NzU0ZTMzZWY1NWI3OTFlM2FmZDhiZjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-X52-e1bwzbsDWH15WpUnEiTmfY9d198tlaEAsw-Cb8) obsolete.

cc @Fabrizio69

Lawproto, 2024-05-23:
@Fabrizio69 I checked, I please ask you to do this:
- Substitute `17845` for `4591` inside this BoM:
[image]
- Delete `17845`.

Everything will fall back into place.

closed 2024-05-23 by Lawproto

Fabrizio69, 2024-05-27:
Done: deleted `cod.17845`

Miche19, 2024-05-27:
@Mick3Lozzo @Lawproto after a check, only 20 pieces of the code 4591 are present in the warehouse

Fixed real quantities in WinGst

cc: @Fabrizio69