mesh-iit/cad-libraries #174: Update Prepare-PTC-Creo-Mechanism-for-URDF  to refer to creo2urdf ?

traversaro, 2024-03-21:
## Description of the task

The documentation in https://github.com/icub-tech-iit/cad-libraries/wiki/Prepare-PTC-Creo-Mechanism-for-URDF still refer to simmechanics-to-urdf, probably we should update it to refer to creo2urdf instead? 

fyi @Nicogene @mfussi66 @marcellosavoldi

mfussi66, 2024-03-21:
Along with it, I suggest a session with the mechies to install and try out creo2urdf together, to answer to doubts, spot bugs and problems (such as https://github.com/icub-tech-iit/creo2urdf/issues/84)

marcellosavoldi, 2024-03-22:
@Mick3Lozzo , would you feel up to editing the Wiki section with the proposed updates?
... it seems to me that you made it?
@mfussi66 , could you collaborate on the drafting of the document?
... I have not exported URDF for a long time ...

cc: @fiorisi

closed 2026-02-04 by Lawproto