mesh-iit/cad-libraries #176: ECAD - PICOBLADE-6X1-MALESTR-P125 - 1388 - PCB connector, Picoblade, 6 contacts, Straight

Lawproto, 2024-04-03:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [link or attachment ] *               |       |
| Manufacturer info  | [Manufacturer name + serial number] * |       |
| Alias Creo         | [Alias or filename MCAD] \**          |       |
| MCAD Contact       | Lorenzo Protopapa, @Lawproto       |       |
| ECAD Contact       | Francesco Garini, @fgarini              |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [x] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [x] Added assignees (min. 1 mechanical + 1 electrical contact).
- [x] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

Lawproto, 2024-04-03:
@fgarini can you please provide me with a screenshot for the pin 1 position? You can find the alias of the component in title.

marcellosavoldi, 2024-04-03:
cc @maurizbo

fgarini, 2024-04-03:
[image]

Lawproto, 2024-04-03:
@fgarini I am understanding the origin is on the vertical axis of symmetry of the whole component and on the horizontal axis of symmetry of pads 7 and 8. Correct?

marcellosavoldi, 2024-04-03:
all Picoblades made so far have the origin in the centre of the PAD of PIN 1.... are you sure you want to make it different?

[image]

maurizbo, 2024-04-03:
I agree with @marcellosavoldi, unfortunately this code has already been used in lots of previous cards (like 2FOC, AMO1 .., lastly in amcx-jig) with the old way to choice the origin,  in the center of the component. In this case the center has been individuated in the middle between the two "retention" pins.
That's all folks  😒

Lawproto, 2024-04-03:
@maurizbo so be it. Thank you all.

Lawproto, 2024-04-08:
> * Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).

[image]