mesh-iit/cad-libraries #179: Vision Light Tech - ACF125B0158IRM12 - S-Mount lens

Lawproto, 2024-04-11:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | Eyes Mk4  |
| Alias Creo           | VLT_ACF125B0158IRM12              |
| IIT-Code             | [if available]                         |
| Applicant            | Lorenzo Protopapa, @Lawproto                 |
| Supervisor           | Nicolò Genesio, @Nicogene                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Lawproto, 2024-04-11:
I created the item following the datasheet.

[image]

Not sure if the `LE` category is correct for this item. `COMM` and `ELTR` also seem possible to me.

Lawproto, 2024-04-12:
For now I added the same mass of the Basler. To be checked when we receive the physical objects.

marcellosavoldi, 2024-04-15:
> I created the item following the datasheet.
> 
> 
> Not sure if the `LE` category is correct for this item. `COMM` and `ELTR` also seem possible to me.

The LE category in CAD is not usually used.
Better to use another category... like COMM .
Honestly ELTR I would only use it for components to be used on electronic boards (to be used on CAD-Collaboration).
what do others think?

Lawproto, 2024-04-15:
@marcellosavoldi `ELTR` or `COMM` makes a huge difference because it establishes whether the item is gonna be filtered from the mechanical BoM or not. I can see the reasoning beyond the `ELTR` since they are usually mounted on a board module.

For instance, the Basler camera (4k head) has been categorized as `ELTR`, so I am assuming it appears somewhere in the electronic BoM. Is that so?

[image]
 
cc @maggia80

maggia80, 2024-04-15:
@Lawproto, yes we are used to including the lens in the Logic Schematics. We will add them in the BoM of the head logic

Lawproto, 2024-04-15:
@maggia80 so `ELTR` be it. I am assuming the related board modules follow the same category.

maggia80, 2024-04-15:
yes!

marcellosavoldi, 2024-04-15:
Perfect... as long as it is present in the electronic BOM, because Wingst the ELTR category skips it.
great!

Lawproto, 2025-01-16:
And to quote Little Red Riding Hood's Huntsman before slaying the Big Bad Wolf: "Long have I sought you." Closing.

[image]

closed 2025-01-16 by Lawproto