mesh-iit/cad-libraries #183: Faulhaber - 3216X024BXTH - Electric motor 24V

Lawproto, 2024-05-16:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | ergoCub neck    |
| Alias Creo           | 	FAULHABER_3216X024BXTH                        |
| IIT-Code             | [if available]                         |
| Applicant            | Lorenzo Protopapa, @Lawproto                 |
| Supervisor           | Nicolò Genesio, @Nicogene                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Lawproto, 2024-05-17:
The item is ready with the mass of the datasheet, to be updated when the physical object will be available.

[image]

marcellosavoldi, 2024-05-20:
Hi @Lawproto ,
it might be useful to enter threads, axes and it might be useful to run a 'recognition pattern' ...
What do you think?
we usually do this and many times it is useful when assembling...

Lawproto, 2024-05-20:
@marcellosavoldi sounds good, can you offer an example so I can copy?

Lawproto, 2024-05-20:
We agreed T2T to check `FAULHABER_3216X024BXTH` for reference.

Lawproto, 2024-05-21:
I seized the day and added the info coming from the datasheet into annotations, as well.

[image]
[image]
[image]

I like the idea of having all the useful information in the 3D. Who knows where this may bring us in the future.
cc @icub-tech-iit/silo-mech

Lawproto, 2024-06-07:
> I have to update the string `3216X024BXTH` to `3216W024BXTH` everywhere.

✅ CAD updated.

Lawproto, 2024-08-28:
Mass updated. I was hoping for a `666` but what can you do about that. Closing.

[image]

closed 2024-08-28 by Lawproto