mesh-iit/cad-libraries #188: ECAD - [MOLEX_46437-9321] - [tbd] - [short description]

marcellosavoldi, 2024-06-10:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [link](https://www.molex.com/en-us/products/part-detail/464379321)  |       |
| Manufacturer info  | Molex 46437-9321 |       |
| Alias Creo         | MOLEX_46437-9321   |       |
| MCAD Contact       | Marcello Savoldi  ,   @marcellosavoldi      |       |
| ECAD Contact       | Andrea Solari, @andresoll (cc @Maurizio Borgagni, @maurizbo )  |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

[image]

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")
MOLEX_46562-9321

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2024-06-10:
https://github.com/icub-tech-iit/mechatronics/issues/87

maurizbo, 2024-06-19:
I'd suggest to change the alias from 
MOLEX_46437-9321
to
MOLEX_EXT_TEN60_15PIN

cc @marcellosavoldi @andresoll

maurizbo, 2024-06-20:
EXTreme Ten60Power Right-Angle Plug with Side Guides, 1 DC Power Circuit, 15 Signal Circuits, 1 DC Power Circuit, Press Fit Tail

@marcellosavoldi All it's seemed to me too much long

This is a kind of component that will not be used anymore ...so could be useful to avoid similarity

marcellosavoldi, 2024-06-25:
the [procedures](https://github.com/icub-tech-iit/cad-libraries/wiki/Create-a-ECAD-component-in-CREO) read as follows:
`The file name must be decided in accordance with the electronic team.`
... so we can proceed as you think best.

marcellosavoldi, 2024-07-05:
Connector renamed as required

> For The final name of the MOLEX 46437-9321
> 
> MOLEX_EXT_TEN60_15PIN

marcellosavoldi, 2024-07-09:
@maurizbo , @andresoll ,
can we close this issue?

maurizbo, 2024-07-09:
Hi @marcellosavoldi  can you wait just the quick check from @andresoll in the pads figure ? We can close tomorrow

marcellosavoldi, 2024-07-11:
@maurizbo , @andresoll , @fgarini 

about the comment [HERE](https://github.com/icub-tech-iit/electronics-boards/issues/443#issuecomment-2220821219)
Can you confirm the new ALIAS name " **MOLEX_EXT_TEN60_15PIN_MALE_90** " for the Molex code 46437-9321 ?

marcellosavoldi, 2024-07-11:
@maurizbo ,
if you want check your model Vs the CAD model , following some dimensions:

[image]

marcellosavoldi, 2024-07-12:
SOME DIMENSIONS TO CHECK THE PADS LIBRARIES FIGURE:
[image]

fgarini, 2024-07-12:
## figure to adjust
(it's not symmetrical)
[image]

[image]
Check consequently:
[image]
[image]

valegagge, 2024-07-22:
Hi guys,
I saw there are still 2 tasks to complete this task. Is it correct or can we close the issue?

closed 2024-07-22 by marcellosavoldi