mesh-iit/cad-libraries #189: ECAD - [PICOBLADE-7X1-MALESTR-P125] - 3947] - [Picoblade 7x1]

marcellosavoldi, 2024-07-02:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [[MOLEX 53398-0771](https://www.molex.com/en-us/products/part-detail/533980771#mates-with-use-with) ] *               |       |
| Manufacturer info  | MOLEX 53398-0771 |       |
| Alias Creo         | PICOBLADE-7X1-MALESTR-P125 \**          |       |
| MCAD Contact       |     Marcello Savoldi, @marcellosavoldi          |       |
| ECAD Contact       | Maurizio Borgagni, @maurizbo               |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

[image]

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

51021-0700

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2024-07-02:
@maurizbo , Can you check the origin position on PADS?

maurizbo, 2024-07-03:
[image]

maurizbo, 2024-07-03:
I changed the origin of this connector because it was in a strange positions. The cards where it is used are only test equipment or not  icub tech card. Anyway in the past it has been used here

[image]

marcellosavoldi, 2024-07-05:
Se La figura PADS è ok si può chiudere.

Confermi @maurizbo ?

closed 2024-07-05 by marcellosavoldi