mesh-iit/cad-libraries #190: Editing HFUC-5-50-2A-R-SP_B- Improvements in the grain area

MeliTaliani, 2024-07-03:
| Description   | Field                              |
|:--------------|:-----------------------------------|
| Category      | MCAD                       |
| Alias Creo    | HFUC-5-50-2A-R-SP_B                  |
| IIT-Code      | 2708                     |
| Applicant     | Melissa Taliani , @MeliTaliani         |
| Supervisor    | Lorenzo Protopapa, @Lawproto            |
| Maintainer    | Marcello Savoldi, @marcellosavoldi |
| Coworkers     | @icub-tech-iit/silo-mech           |

## Description and motivation of the changing request: <br>

Changes in the area following the observation of an oversized grain hole.

[image]

## Checklist:
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Checked that the associated [WINGST](http://wingst.icub.iit.local/) code is up-to-date and that the manufacturer and at least one supplier are defined.

Lawproto, 2024-07-03:
@MeliTaliani I am available in case you need help integrating the set screw geometry inside the library component.

MeliTaliani, 2024-07-03:
Size of the set screw hole reduced.

[image]

Lawproto, 2024-07-04:
@MeliTaliani good job! Did you also manage to integrate the set screws as features inside the part?