mesh-iit/cad-libraries #192: ECAD - [	MINIFIT-2X2-FEM90-P420 ] - [11285] - [Mini-Fit Jr. Header - 9° - 2x2 ]

marcellosavoldi, 2024-08-14:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [LINK  ](https://www.molex.com/en-us/products/part-detail/26013128)             |       |
| Manufacturer info  | Molex  26013128|       |
| Alias Creo         | MINIFIT-2X2-FEM90-P420         |       |
| MCAD Contact       | Marcello Savoldi, @marcellosavoldi          |       |
| ECAD Contact       | Maurizio Borgagni , @maurizbo             |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

[image]

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

[Molex 39013048](https://www.molex.com/en-us/products/part-detail/39013048)

### Notes (optional)

Già esistente su Wingst, mancano però tutte le figure

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

valegagge, 2024-10-29:
@marcellosavoldi @maurizbo news on this task?

valegagge, 2024-11-08:
Don't use anymore.

closed 2024-11-08 by valegagge