mesh-iit/cad-libraries #199: ECAD - [PCA9615DPZ] - [18258] - [component for I2C ext]

marcellosavoldi, 2024-10-23:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | [NXP - PCA9615 I2C bus extender.pdf](https://github.com/user-attachments/files/17488023/NXP.-.PCA9615.I2C.bus.extender.pdf)
              |       |
| Manufacturer info  | NXP semiconductor PCA9615DPZ |       |
| Alias Creo         | PCA9615DPZ        |       |
| MCAD Contact       | Marcello Savoldi , @marcellosavoldi                |       |
| ECAD Contact       | Maurizio Borgagni , @maurizbo            |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

@maurizbo ,  can you added the image here?

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2024-10-23:
[image]

[image]

marcellosavoldi, 2024-10-23:
I checked the other similar compionents... the origin is in the middle...is for this the same?
[image]

maurizbo, 2024-10-24:
Hi @marcellosavoldi sorry for the delay, here you are the 0 rotation of the U1 component (origin in the center).

[image]

cc @andresoll

marcellosavoldi, 2024-10-25:
done
[image]

closed 2024-10-25 by marcellosavoldi