mesh-iit/cad-libraries #204: [manufacturer] - [part number] - [short description]

Elio75, 2025-02-04:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  |  [datenblatt_63909.pdf](https://github.com/user-attachments/files/18654190/datenblatt_63909.pdf) / [[dual-band-wireless-ac-8265-brief-1219566.pdf](https://github.com/user-attachments/files/18654122/dual-band-wireless-ac-8265-brief-1219566.pdf) |
| Project              | [ergoCub]      |
| Alias Creo           | [DELOCK_62858] - [INTEL_8265NGWMG]     |                |
| IIT-Code             | [13463] - [13462]                      |
| Applicant            | Elio Massa @Elio75                     |
| Supervisor           |  F. Larosa, @Fabrizio69                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Elio75, 2025-02-04:
## I designed these boards because a 3D model cannot be found ##

DELOCK_62858 Adattatore Delock Mini PCIe > M.2 Key E Slot

-The **DELOCK_62858** board is a kit, i.e. board plus fixing hardware, so I drew it representing board plus mounting hardware

[image]

---

**INTEL_8265NGWMG** Intel 8265, Mini PCI-e WiFi/802.11 Modules Dual Band Wireless-AC 8265, 2x2 AC + BT 4.2, M.2 KEY E

[image]

fiorisi, 2025-02-04:
Great @Elio75. Please also take a picture of the parts on the weighing scale, ensuring that the displayed weight is clearly visible.