mesh-iit/cad-libraries #219: DIN 7979/D  UNI 6364/B  ISO 8735 Family table - Cylindrical pin with female thread, steel plain, hardened, tolerance class m6

Mick3Lozzo, 2025-02-19:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [Wurth manufacturer page](https://eshop.wuerth.it/Categorie-di-prodotti/DIN-7979-acciaio-temperato-grezzo/31143510100201.cyid/3114.cgid)                 |
| Project              | ergoCub and others    |
| Alias Creo           | ISO8735                       |
| IIT-Code             | various instances related                       |
| Applicant            | Michele Gesino, @Mick3Lozzo               |
| Supervisor           | Luca Fiorio, @fiorisi               |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

Some other datasheets found on the web (anyway there is a UNI/ISO normative related)

[DIN 7979-D  UNI 6364-B  ISO 8735.pdf](https://github.com/user-attachments/files/18870365/DIN.7979-D.UNI.6364-B.ISO.8735.pdf)
[DIN 7979-D  UNI 6364-B  ISO 8735 bis.pdf](https://github.com/user-attachments/files/18870364/DIN.7979-D.UNI.6364-B.ISO.8735.bis.pdf)

### Notes:

This is a new ISO family table.

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

Mick3Lozzo, 2025-04-23:
We haven't ever bought these pins, but the family table has been created. We could close this issue @marcellosavoldi ?

marcellosavoldi, 2025-04-23:
if the object lives in CAD-Libraries I would handle everything as we usually do even if we do not purchase
or maybe I didn't understand the question?

Mick3Lozzo, 2025-04-23:
The object lives in CAD libraries. I created the family table, but in the end, we didn't purchase these items.

Mick3Lozzo, 2025-04-24:
> if the object lives in CAD-Libraries, I would handle everything as we usually do, even if we do not purchase, or maybe I didn't understand the question?

So we can close the issue, I think.

closed 2025-04-24 by Mick3Lozzo