mesh-iit/cad-libraries #222: ECAD - components for BP-SPI-LED Electronic Board

marcellosavoldi, 2025-02-28:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | see table  *             |       |
| Manufacturer info  | see table |       |
| Alias Creo         | see table **          |       |
| MCAD Contact       | Marcello Savoldi, @marcellosavoldi               |       |
| ECAD Contact       | Maurizio Borgagni, Andrea Solari               |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

| # | IIT-CODE |           ALIAS           |                              Description                              |             CAD figure infos            | Link |
|:-:|:--------:|:-------------------------:|:---------------------------------------------------------------------:|:---------------------------------------:|------|
| 1 | 18565    | LED-RGB-WS2812B           | LED-RGB-WS2812B                                                       | existing                                | [Link](https://www.adafruit.com/product/4684) |
| 2 | 18618    | SMT1-01-Z                 | Surface Mount   Tactile Switches                                      | change file name (nidec-comp_smt1-01-z) | [Link](https://www.mouser.it/ProductDetail/Nidec-Components/SMT1-01-Z?qs=XeJtXLiO41Qip7f%252BPNPtew%3D%3D) |
| 3 | 18631    | BZT52-B12-QX              | Diode Zener,   Pv: 590mW, Vz: 12V, If: 250mA, SMD, SOD-123-2          | to do                                   | [Link](https://www.mouser.it/ProductDetail/Nexperia/BZT52-B12-QX?qs=rQFj71Wb1eVMbK8ZYprizw%3D%3D) |
| 4 | 16955    | 16955                     | FUSE BOARD   MOUNT 10A 32VAC 63VDC SLOW BLOW (SMD 1206)               | wrong ALIAS                             | [Link](https://it.rs-online.com/web/p/fusibili-non-ripristinabili/7032770?searchId=e5d14cca-60ba-4181-b3bc-b4466bea5bd8&gb=s) |
| 5 | 18632    | 18632                     | Dual N- P   Channel Mosfet, Vds=60V, Id=6.5A, 7A, Rds=32mR, SMD, SOP8 | wrong ALIAS                             | [Link](https://www.mouser.it/ProductDetail/ROHM-Semiconductor/SH8MC5TB1?qs=QNEnbhJQKvamigFoyVNsNg%3D%3D) |
| 6 | 1124     | PICOBLADE-6X1-MALE90-P125 | PICOBLADE-6X1-MALE90-P125                                             | existing                                | [Link](https://it.farnell.com/molex/53261-0671/connettore-basetta-6pos-1fila/dp/1125375) |
|   |          |                           |                                                                       |                                         |      |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

KathrynAdams, 2025-04-01:
Ciao @andresoll, can you please confirm the position of origin of the following components: 

| # | IIT-CODE |           ALIAS           |
|:-:|:--------:|:-------------------------|
| 1 | 18631    | BZT52-B12-QX              |
| 2 | 16955    | FUSE-1206-10A-32VAC-63VDC                     |
| 3 | 18632    | SH8MC5TB1                    |

andresoll, 2025-04-01:
Hi @KathrynAdams ,

yes sure:

[image]

[image]

we will not use the fuse in the final version. 
Check the new bom below.

## Part Lister Output for Board1  
**Generated on Tuesday, April 01, 2025**  

| #  | QTY | Part Number        | Description                                |
|----|-----|--------------------|--------------------------------------------|
| 1  | 1   | LED-RGB-WS2812B    | Led diode, RGB, SPI, 12mA, 5V SMD 2.2x2.0mm |
| 2  | 1   | CAP-0201-X5R-100NF-6V3-10TOL | Multilayer Ceramic Capacitor, 100nF, 6V3, 10%, X5R, SMD, 0201 |
| 3  | 1   | BZT52-B12-QX       | Diode Zener, Pv: 590mW, Vz: 12V, If: 250mA, SMD, SOD-123-2 |
| 4  | 1   | PICOBLADE-6X1-MALE90-P125-0671 | PCB connector, Picoblade, 6 contacts, 1 row, male, p.1.25mm, 90°, w to b, nolock, key, SMD |
| 5  | 1   | SH8MC5TB1          | Dual N-P Channel Mosfet, Vds=60V, Id=6.5A, 7A, Rds=32mR, SMD, SOP8 |
| 6  | 1   | RES-0402-100R-0TOL1 | Thin film precision resistor, 100R, 0.063W, 0.1%, SMD, 0402 |
| 7  | 1   | RES-0402-22K-1TOL  | Resistor Thick film, 22K, 0.063W, 1%, SMD, 0402 |
| 8  | 1   | RES-0603-10R-1TOL  | Resistor Thick Film, 10R, 0.1W, 1%, SMD, 0603 |
| 9  | 1   | RES-0603-33K-1TOL  | Resistor Thick film, 33K, 0.1W, 1%, SMD, 0603 |
| 10 | 1   | SMT1-01-Z          | Switch button, 7.5x6mm, SMD, 50mA, 48V     |

marcellosavoldi, 2025-04-02:
@KathrynAdams , also check the list of tables created by Andrea, if we need to create more parts we can start doing that.
The RES-0xxxx are in the following familytables: RES-0402.prt and RE-0603.prt.

PS: If you also want to display the variants of the familytable in the Creo folder, you must activate the display ( I recommend that you then remove it )

[image]

KathrynAdams, 2025-04-02:
All parts in the table created by Andrea have already been created. 
I can close the issue