mesh-iit/cad-libraries #225: ECAD - [ALIAS] - [IIT-CODE] - [short description]

marcellosavoldi, 2025-04-11:
| Description        | Field                                 | Notes |
|:-------------------|:--------------------------------------|:------|
| Datasheet          | see notes               |       |
| Manufacturer info  | several Manufacturer name |       |
| Alias Creo         | see list         |       |
| MCAD Contact       | @marcellosavoldi , @KathrynAdams         |       |
| ECAD Contact       | Francesco Garini        |       |
| Maintainer         | Marcello Savoldi, @marcellosavoldi    |       |

Legend:  
 - "\*" = To compile only if this information are missing in the database (Wingst)  
 - "\**" = To compile only if it is a preliminary name

### Image with origin e PIN 1 position by PADS (mandatory)

severals

###  Attachments (optional)

### Plug connector information (mandatory for connectors "board to wire")

### Notes (optional)

the following request was made for the following components to be made for the BAT circuit board:

- [ ] AT24CM01
- [ ] TERMINAL-TEST-S1751-1206
- [ ] LED-0603-RED-2V-40MCD
- [ ] LED-0603-ORANGE-2V-40MCD
- [ ] MAX3051
- [ ] TXS0104ERGYR
- [ ] CM-FILTER-7X4-330UH-0A3-80VDC
- [ ] TPS2493
- [ ] RES-1224KELVIN-OR002-1TOL
- [ ] LM5575MH
- [ ] LM5088MH
- [ ] MMSZ4702T1G
- [ ] IND-5X5-33UH-820MA-10TOL
- [ ] LM5050-1

### Checklist:
- [ ] Added IIT-CODE from [WINGST](http://wingst.icub.iit.local/).
- [ ] Added assignees (min. 1 mechanical + 1 electrical contact).
- [ ] Added Link/datasheet/drawing or others information (or information into Wingst database compiled).
- [ ] Added details about the PIN 1 and the orientation of the origin of the component (img/sketch/print-screen).
- [ ] Added details about the PLUG connector (added datasheet, or Wingst code).
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).

marcellosavoldi, 2025-04-28:
@KathrynAdams , can you help me to finish?

@maurizbo can you  help me and Katie with the origin and position about all figures above?

maurizbo, 2025-05-06:
res-1224kelvin-0r002-1tol

[image]

Origin in the center
-----------------------------------------------------
LM5575MH & LM5088MH

[image]

Origin in the center
-----------------------------------------------------
MMSZ4702T1G

[image]

Origin in the center

-----------------------------------------------------

IND-5X5-33UH-820MA-10TOL