mesh-iit/cad-libraries #228: [Synapticon] - [ACTILINK-JD8, ACTILINK-JD10, ACTILINK-JD12] - [revolute motors with planetary reducers]

KathrynAdams, 2025-04-18:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [link or attachment ]                  |
| Project              | [where will the component be used]     |
| Alias Creo           | [maker + code]                         |
| IIT-Code             | [if available]                         |
| Applicant            | Kathryn Adams, @KathrynAdams                 |
| Supervisor           | Name Surname, @username                |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes:

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

marcellosavoldi, 2025-04-29:
Created the 3 joints:

[image]

[image]

all datas:

[image]

marcellosavoldi, 2025-05-05:
we have to create the Wingst code ...

marcellosavoldi, 2025-05-13:
[image]

closed 2025-05-13 by marcellosavoldi

marcellosavoldi, 2025-08-20:
# Thread with shallow depth (on JD12).

[image]

[image]