mesh-iit/cad-libraries #247: KOLLMORGEN - TBM2G-05008C-ANAA-00 - Frameless motor

fiorisi, 2025-07-22:
| Description          | Field                                  |
|:---------------------|:---------------------------------------|
| Datasheet component  | [Datasheet](http://wingst-icub.iit.local/direct_download.php?LINK=%2F%2Fiiticubstor01.iit.local%2Ficub-tech%2Fmechatronics%2FDatasheet%2FMotors%2FKollmorgen%2FTBM2G-05013C-ANAA-00.pdf)                  |
| Project              | ergoCub lowerbody     |
| Alias Creo           | KOLLMORGEN_TBM2G-05013C-ANAA00                         |
| IIT-Code             | 18657                         |
| Applicant            | Luca Fiorio, @fiorisi                 |
| Supervisor           | Marco Maggiali, @maggia80                 |
| Maintainer           | Marcello Savoldi, @marcellosavoldi     |
| Coworkers            | @icub-tech-iit/silo-mech               |

### Attachment:

### Notes: WIP

### Checklist:
Complete the checklist in order to proceed with the request.

- [ ] Created the Alias following [the standard convention](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#commercial-components-coding-standard).
- [ ] Added, if necessary, the source files (STEP, IGS, etc), the links/datasheets/drawings or other information used to create the library part.
- [ ] File imported using the Creo [start part commercial](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#standard-part-for-commercial-components).
- [ ] Checked [mass properties](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#mass-properties-of-commercial-parts).
- [ ] Checked if one [special case](https://github.com/icub-tech-iit/cad-libraries/wiki/Mechanical-design-guidelines#special-cases-and-examples) applies to this component.
- [ ] Converted the files with the [EDU-COM license](https://github.com/icub-tech-iit/cad-libraries/wiki/PTC-Creo-Guidelines#save-file-with-commercial-license).
- [ ] Created IIT code in [WINGST](http://wingst.icub.iit.local/) with manufacturer and at least one supplier.

fiorisi, 2025-07-22:
## ROTOR

[image]

[image]

maggia80, 2025-08-12:
Temperature sensor/PTC: **TP1000**

cc: @valegagge @andresoll @aazocar7